IRNAS/KORUZA-router #1: Processor choice (MT7621 or different...)

SloMusti, 2016-06-16:
Choosing the core processor is key to cover all features, the choice is somewhat limited to Mediatek chips that are well supported already no no significant work on supporting them in openwrt is required.

MT7621: https://wikidevi.com/wiki/MediaTek_MT7621

* MT7621AT
  * 5xGbE, 1xRGMII, 3xPCIe, 1xUSB 3.0, 2xUSB 2.0,
  * L1-32/32kB, L2-256kB, DDR2/DDR3 256/512MB, LFBGA346
* MT7621ST
  * 5xGbE, 1xRGMII, 3xPCIe, 1xUSB 3.0, 2xUSB 2.0,
  * L1-32/32kB, L2-256kB, DDR2/DDR3 256/512MB
* MT7621NT
  * 2xRGMII, 2xPCIe, 1xUSB 3.0, 2xUSB 2.0,
  * L1-32/32kB, L2-256kB, DDR2 256MB

VojislavM, 2016-11-16:
Processor is Mediatek [MT7621AT](https://wikidevi.com/wiki/MediaTek_MT7621), it is the main processor on [WiTi board](https://mqmaker.com/product/witi-board/)

closed 2016-11-16 by VojislavM

SloMusti, 2016-11-16:
MT7621NT is not suitable due to the lack of internal switch

VojislavM, 2016-11-16:
Ideas for the board and processor: https://forum.openwrt.org/viewtopic.php?id=59092&p=2