IRNAS/KORUZA-router #10: Power supply

SloMusti, 2016-06-16:
The board must be powered primarily via PoE for KORUZA application, but should have input for AC/DC power in the secondary application.

Power input must be sufficient to power unit itself with KORUZA periphery as well as one more router and KORUZA unit through PoE output ports.

SloMusti, 2016-06-16:
Power input DC jack should be on the board or at least a pad for adding it later on interchangeably with a screw terminal connector.

VojislavM, 2016-11-17:
[image]

SloMusti, 2016-11-17:
Power input to the router will be 5V regulated, external power board is used for PoE. Above connector is placed on the router to connect to an external power board.

VojislavM, 2016-11-17:
Power connector V2, ground is common.
[image]