IRNAS/KORUZA-router #4: USB port

SloMusti, 2016-06-16:
USB port 2.0 is required. One port for webcam one port for control board in KORUZA specific application. There is also USB 3.0 option on MT7621 which could be implemented, but likely not required in given specifications. If cost is low 3.0 port should be available outside and 2.0 port internally on GPIO header or some socket as on MqMaker WiTi.

Implementation depends on #1

VojislavM, 2016-06-16:
USB connection could be done with dual USB connector. This way we would save space on the board, and if the board is used outside of the Koruza unit, we will still have two USB connectors, which is good for the development. 
[image]

We should pay attention to the footprint of the USB connector, so it could be replaced with the differnt connector when a board is placed inside of the Koruza unit.

SloMusti, 2016-11-16:
We will use one external USB 3.0 port facing the same way as ethernet ports and one internal on the expander board.