IRNAS/KORUZA-router #21: Determine why the WAN port doesn't work

kostko, 2016-11-24:
It could be either the PHY or the RGMII support in the Linux kernel as a proprietary driver from MediaTek is required for this port to work (for OpenWrt there is [this heavily modified fork](https://forum.mqmaker.com/t/alternative-openwrt-mtksdk-build-for-the-witi-board-wip/272) where and old kernel version is used to get the proprietary module to load).

We should ask the MQmaker guys if they know what exactly is the problem (we may be able to change the PHY).

SloMusti, 2016-12-01:
They are claiming the Mediatek driver needs to be simply ported to openwrt for RGMII port and that is it.

kostko, 2016-12-01:
Do they have the source code? Or how do they think it "needs to be simply ported"?